NetFPGA/P4-NetFPGA-public #2: SUME drivers error

zjuan22, 2018-06-04:
We are following the P4-Netfga wiki tutorial (https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Getting-Started).  When we run the step 9) Build and load the SUME drivers.  The console shows an error message as follow:

make -C /lib/modules/4.13.0-43-generic/build M=/home/weihen/workarea/netfpga/P4-NetFPGA/lib/sw/std/driver/sume_riffa_v1_0_0 modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-43-generic'         
 CC [M]  /home/weihen/workarea/netfpga/P4-NetFPGA/lib/sw/std/driver/sume_riffa_v1_0_0/sume_riffa.o
 scripts/Makefile.build:323: recipe for target '/home/weihen/workarea/netfpga/P4-NetFPGA/lib/sw/std/driver/sume_riffa_v1_0_0/sume_riffa.o' fai    led
 Makefile:1550: recipe for target '_module_/home/weihen/workarea/netfpga/P4-NetFPGA/lib/sw/std/driver/sume_riffa_v1_0_0' failed
 make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-43-generic'          
 Makefile:68: recipe for target 'modules' failed

Can you help us, please?

sibanez12, 2018-06-05:
A couple questions:
1. Which step fails? `# make all` or `# make install`?
2. Are you running the commands as the root user?
3. What linux distribution are you using? `# lsb_release -a`
4. What version of gcc are you using? `# gcc --version`
I've tested with Ubuntu 14.04, kernel v4.13.8, and gcc 4.9.4. If you're using a newer version of Ubuntu or gcc, I'm wondering that would cause problems.

zjuan22, 2018-06-05:
Thanks for your fast response, 

1) it fails in #make all, also in #make install.
2) Yes, we are in sudo user
3) We have ran: lsb_realease -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:printing-9.20160110ubuntu0.2-amd64:printing-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Description:	Ubuntu 16.04.4 LTS
Codename:	xenial
4) or version of is gcc 5.4.0.

We are using a newer version of Ubuntu. Do you recommend install ubuntu 14.04 to work with the Netfpga? There are other repos of sume_riffa driver in Github compatible with latest versions of Ubuntu? 

Cheers,

tudang, 2018-06-05:
Hi,

The sume_riffa driver may fail because the kernel API has changed recently.
Try to modify line 852 of this file lib/sw/std/driver/sume_riffa_v1_0_0/sume_riffa.c<https://github.com/usi-systems/P4-NetFPGA-live/commit/416ff6e2d518cf71dbe7b4510441db14ce17fb1e#diff-3fb44e7306358bc572f8665bba06e2a6>  as follow and rebuild the driver.

/* Statistics. Everyone likes numbers. */
                -static struct rtnl_link_stats64 *
                +static void
                 sume_get_stats64(struct net_device *netdev, struct rtnl_link_stats64 *storage)
                 {

Tu

tudang, 2018-06-05:
I missed the following change:
line 887 in the same file.
- return (storage);

Tu

On Jun 5, 2018, at 8:37 PM, Tu Dang <huynh.tu.dang@usi.ch<mailto:huynh.tu.dang@usi.ch>> wrote:

Hi,

The sume_riffa driver may fail because the kernel API has changed recently.
Try to modify line 852 of this file lib/sw/std/driver/sume_riffa_v1_0_0/sume_riffa.c<https://github.com/usi-systems/P4-NetFPGA-live/commit/416ff6e2d518cf71dbe7b4510441db14ce17fb1e#diff-3fb44e7306358bc572f8665bba06e2a6>  as follow and rebuild the driver.

/* Statistics. Everyone likes numbers. */
                -static struct rtnl_link_stats64 *
                +static void
                 sume_get_stats64(struct net_device *netdev, struct rtnl_link_stats64 *storage)
                 {

Tu

mharris-t, 2021-03-12:
Hi,

I get the following error when installing the SUME drivers. Is there anything that can be done?

[image]

juraul1, 2021-12-01:
Hi @mharris-t, I have the exact same issue. Did you find a solution? 
sume_riffa driver seems not to work with that kernel version.

mharris-t, 2021-12-01:
@juraul1 Nope. I also basically faced the issue of NetFPGA not getting detected by my system's motherboard. And since only limited amount of motherboards detect NetFPGA So I had to scrap the idea of using it and went onto work with BMv2 unfortunately.

mharris-t, 2021-12-01:
@jrual

mharris-t, 2021-12-01:
@juraul1 

I recommend you to search this repository (link provided below). The repo owner documented the installation of NetFPGA. And some of the error messages can be queried in GitLab and you may find a solution to your issue. The repo helped me solve few of the issues I faced while installing SUME drivers. I am sorry that I cannot help you much, it's been some time since I finished my master's thesis and I do not remember much 😅

https://gitlab.ethz.ch/nicosc/master-thesis

juraul1, 2021-12-01:
@mharris-t yeah I just checked, my motherboard unfortunately do not recognize the board either. 
Thanks for the answer and for the link! :)