NetFPGA/P4-NetFPGA-public #7: Cannot run `vivado_sim_waveform.bash`

AnotherKamila, 2019-02-08:
I am following the `switch_calc` tutorial assignment. I checked out the `master` branch of this repository.

In step 5, I can run `vivado_sim.bash` successfully, and it passes. However, when trying `vivado_sim_waveform.bash`, I get this error:

```
ERROR: [XSIM 43-3225] Cannot find design unit work.glbl_sim in library work located at xsim.dir/work.
```

`glbl_sim` really seems to be missing, there is a `glbl.sdb` but no `glbl_sim`:
```
$ ls xsim.dir/work/*glbl*
xsim.dir/work/glbl.sdb
```

Is this a bug?

Thanks!

AnotherKamila, 2019-02-09:
For the record, still true in the `vivado-2018.2` branch.

sibanez12, 2019-02-09:
Thanks for catching and reporting this. Seems like a minor compatibility issue between SDNet 2018.2 and Vivado 2018.2. Just pushed a temporary fix to the vivado-2018.2 branch. Please pull these changes and try again.

AnotherKamila, 2019-04-22:
Sorry for the late response. The change fixes the issue, so I'm closing this.

closed 2019-04-22 by AnotherKamila

cube1us, 2020-11-30:
I ran into this same message, also in 2018.2 (I have not tried it in 2020.1 - am posting here in case someone else runs into this.

Background:  large VHDL project with VHDL test bench.  Functional simulation at the implementation went fine.  When I tried to do a post-implementation timing simulation (after accepting the message warning about timing simulations only being supported in Verilog) it failed with:

ERROR: [XSIM 43-3225] Cannot find design unit work.glbl in library work located at xsim.dir/work.

I first tried un-checking option xsim.elaborate.load_glbl* which did not have any apparent effect.

After reading through this thread, the cure for me was to copy file glbl.sdb from xil_defaultlib into directory work (all under project\project.sim\sim_1\timing\xsim\xsim.dir)

It was curious because the command that was run seemed to indicate that both xil_defaultlib and work were in play:

Running: P:/Pro/Xilinx/Vivado/2018.2/bin/unwrapped/win64.o/xelab.exe -wto 457e1d21cd7d4152894e97e20dab12d7 --incr --debug typical --relax --mt 2 --maxdelay -L xil_defaultlib -L work -L simprims_ver -L secureip --snapshot IntegrationTest2_tb_time_impl -transport_int_delays -pulse_r 0 -pulse_int_r 0 -pulse_e 0 -pulse_int_e 0 work.IntegrationTest2_tb work.glbl -log elaborate.log