NetFPGA/P4-NetFPGA-public #8: Cannot synthesize `switch_calc` example project

AnotherKamila, 2019-02-08:
Hello,

I am trying to complete the `switch_calc` tutorial. The simulation in step 5 passes, so I believe my P4 code is correct. However, step 10 ("Compile the bitstream") fails. I am unsure whether step 9 ("Run the SUME simulation") succeeds or fails -- there is no error and `make` is happy, but there are some CRITICAL WARNINGs above, so it might be failing silently.

I have not changed any files besides `switch_calc.p4` and I believe I did follow the setup and tutorial steps. I am using Vivado 2016.4 and SDNet 2018.2. I tried both with the latest commit from the `master` branch and the `v1.3.1` release.

I am attaching the log from the synthesis: [runme.log](https://github.com/NetFPGA/P4-NetFPGA-public/files/2846191/runme.log).

Any help would be much appreciated!

sibanez12, 2019-02-08:
Try using Vivado 2018.2 & SDNet 2018.2 and use the vivado-2018.2 branch of the P4-NetFPGA-live repo, which will be merged into the master branch soon.

AnotherKamila, 2019-02-09:
Thank you, that works. Happy merging!

closed 2019-02-09 by AnotherKamila