NetFPGA/P4-NetFPGA-public #9: switch_calc tutorial: simple_sume_switch bitfile not generated

nniranjhana, 2019-03-14:
I've been following the instructions for the first tutorial, [the switch calculator](https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Tutorial-Assignments#assignment-1-switch-calculator). However when I am done with step 10, I am not able to find the bitfile `simple_sume_switch.bit` that is supposed to be generated in the directory `$NF_DESIGN_DIR/bitfiles`.

The only bitfile I find here is `switch_calc.bit`.

Please find the make log attached:
[make_log.txt](https://github.com/NetFPGA/P4-NetFPGA-public/files/2965424/make_log.txt)

**Environment:**
Ubuntu 14.04.5 LTS
Floating Server Tools Linux (Flexv11.14.1.0) used with Xilinx

I'm not sure how to check Xilinx version, but it says in the license file
SDNET xilinxd 2019.06
Vivado xilinxd 2019.03

sibanez12, 2019-03-16:
It looks like the bitstream was successfully generated. Did you already run the commands in the step 12?

```
$ mv simple_sume_switch.bit ${P4_PROJECT_NAME}.bit
$ cp $P4_PROJECT_DIR/testdata/config_writes.sh ./
```

That would rename the simple_sume_switch.bit file to switch_calc.bit

nniranjhana, 2019-03-17:
Oh! Okay yes I do have the switch_calc.bit generated. Thanks for clearing up my confusion.

closed 2019-03-17 by nniranjhana