NetFPGA/P4-NetFPGA-public #10: NetFPGA SUME Server Recommendation

ashishkashinath, 2019-03-24:
Hey folks, Do you know of a compatible server configuration (motherboard with appropriate PCIe 3.0 support) with which we can install the NetFPGA SUME board. The thread in digilent https://forum.digilentinc.com/topic/16243-netfpga-sume-server-recommendation/#comment-39284
,redirects to this forum.

Asking since we tried connecting the SUME to a Dell T630 server having a PCIe3.0 slot, which doesn't seem to detect the board in lspci or dmesg.

Update: The board seems to get detected on tty console when connecting a USB cable and logging in using minicom but Xilinx Vivado does not detect the board as a "Hardware Target".

ralfkundel, 2019-03-26:
Hi,

in general:

  * the board in vivado should be detected via USB (JTAG) even if there
    is no PCIe connection. If the board is connected to your local
    computer this should work automatically in vivado. If it is
    connected to a remote server, you have to install xilinx hardware
    server there.
  * for PCIe: you have to:
     1. load a working bitfile on the FPGA (which has a PCIe core,
        connected correctly)
     2. restart the server (without powering off, otherwise the FPGA
        config is lost).
     3. now you should see something in lspci

I had a similar issue with an old computer (which had PCIe 3.0 ).
However, only a 5 Gt/s configuration worked for me. Maybe you should try
this.

regards,
Ralf

-- 
________________________________________________
Ralf Kundel, M.Sc.

Technische Universität Darmstadt
Dept. of Electrical Engineering and Information Technology Multimedia Communications Lab (KOM)
Adaptive Overlay Communications Group (AOC)

Postal address: TU Darmstadt, FB 18, KOM, Rundeturmstr. 10, 64283 Darmstadt, Germany
Location: S3|20 - 216
Tel: +49 6151 16-20855
Email: Ralf.Kundel@kom.tu-darmstadt.de
Web: http://www.kom.tu-darmstadt.de

ashishkashinath, 2019-03-29:
Thanks Ralf  for the steps. 
we were trying out the command :  fpga -f reference_nic.bit to program the FPGA with the bitfile using the XMD tool. 

But we get an error saying "cable is not connected to the host". (when in fact the SUME board is connected to the host on a PCIe3.0_G3 slot as well as is detected via the USB-UART in minicom)

We are using a Dell Poweredge T630 server with a 0NT78X motherboard having a PCIe3.0 slot.

ralfkundel, 2019-03-29:
Hi,
Have you tried to open the hw-manager in vivado gui. So you see the board there?

ashishkashinath, 2019-03-29:
Yes, we tried that, but the board does not get detected as a hardware target.

sibanez12, 2019-03-30:
Here's a [list of PC parts](https://drive.google.com/file/d/143mP0AI9BlX_qMDErX8xjsYjv3FbX3t2/view?usp=sharing) that we're using for [CS344](https://cs344-stanford.github.io/) at Stanford for reference.

If you're seeing the "cable is not connected to the host" error then I'd be suspicious that either: (1) the JTAG cable is not properly connected to the SUME board, (2) the JTAG cable itself is bad/incompatible, or (3) the JTAG connector on the SUME board is broken. (3) is the most cumbersome to repair. You can try looking for broken pins yourself and soldering them or you can submit an RMA to Digilent and ask them to repair for you.