NetFPGA/P4-NetFPGA-public #12: running ./learn.py throws  "socket.error: [Errno 100] Network is down" (Ethernet-Learning-Switch)

hendrikzuellig, 2019-03-29:
I successfully completed the "Assignment 1: Switch Calculator" and "Assignment 3: In-band Network Telemetry (INT)". "Assignment 2: TCP Monitor" was not working for the same reason the Ethernet-Learning-Switch is not working (I believe). When i execute the [last step](https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Ethernet-Learning-Switch)  i get the following error. Same error for Setp 5 from [TCP Monitor Assignment](https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Tutorial-Assignments#assignment-2-tcp-monitor)
Any ideas?
> root@System:~/projects/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects>/learning_switch/sw/learning_sw# ./learn.py
loading libsume..
loading libcam..
Traceback (most recent call last):
  File "./learn.py", line 80, in <module>
    main()
  File "./learn.py", line 76, in main
    sniff(iface=DMA_IFACE, prn=learn_digest, count=0)
  File "/usr/lib/python2.7/dist-packages/scapy/sendrecv.py", line 577, in sniff
    p = s.recv(MTU)
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 485, in recv
    pkt, sa_ll = self.ins.recvfrom(x)
socket.error: [Errno 100] Network is down```

sibanez12, 2019-03-30:
Make sure that the you see the `nf0` interface when you run `ifconfig`. If you don't then try running `# ifdown nf0 && ifup nf0`

hendrikzuellig, 2019-04-01:
@sibanez12 Thank's! It's working now.

nniranjhana, 2019-04-01:
That's great, I'm closing the issue now as it's resolved by @sibanez12 .

hendrikzuellig, 2019-04-03:
I tried to run the TCP Monitor but the `ifdown nf0 && ifup nf0` trick didn't help there. (the learning switch is working tough)

closed 2019-04-15 by hendrikzuellig