NetFPGA/P4-NetFPGA-public #13: Extern for checksum'ing payload

telmich, 2019-04-13:
Hello,

I have found https://github.com/NetFPGA/P4-NetFPGA-public/wiki/IP-Checksum-Extern-Function that describes one extern for computing the IPv4 checksu.

I am working on a P4 NAT64 gateway that heavily relies on computing checksums of udp, tcp, icmp6, which include the payload of the packet (see https://gitlab.ethz.ch/nicosc/master-thesis/blob/master/p4src/checksums.p4).

How would I access the payload in p4-netfpga and/or is there already any extern that already does computations over the headers + payload that could be adopted?

sibanez12, 2019-05-08:
None the P4->NetFPGA externs have access to the packet payload. If you need to access the packet payload for something you have a couple options:

1. You can implement a custom HDL module and insert it before, after, or in between P4-SDNet generated modules, by updating [nf_datapath.v](https://github.com/NetFPGA/P4-NetFPGA-live/blob/master/contrib-projects/sume-sdnet-switch/projects/switch_calc/simple_sume_switch/hw/hdl/nf_datapath.v).
2. P4 programs are translated into Xilinx PX programs. P4 externs are mapped onto PX user engines. User engines can actually receive the packet payload, but that functionality is not exploited by the P4-to-PX frontend. So one option you can explore is to manually edit the generated PX program to pass the packet through your user engine. [Page 50](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_1/UG1012-sdnet-packet-processor.pdf) shows an example user engine with packet ports.

telmich, 2019-05-11:
@sibanez12 thanks for the follow up! As someone who knows a bit about P4, but not much about (V)HDL - do you have some pointers on where to start if I wanted to go down the custom HDL module?

And if I go that way, is there any chance that the code will be accepted upstream?