NetFPGA/P4-NetFPGA-public #16: More detailed "make your own externs" documentation proposal

AnotherKamila, 2019-04-28:
It would have saved me a lot of time if there was documentation about the interface that P4 expects from externs. I propose adding the following paragraph to the wiki, under "Adding your own externs", before the last paragraph:

> Externs must be pipelined: P4 may input data in any cycle, *not* only after the previous computation has finished. In addition to the reset, clock, input, and output signals, there is also an extra input bit `valid_in` and an extra output bit `valid_out`. The `valid_in` bit will be high when we have valid data to compute with (i.e. if it is not set, we can ignore that input). The `valid_out` bit should be set whenever the result of a valid input (i.e. of data that was input when `valid_in` was set) is ready -- this bit signals to P4 that it can take the data and resume execution of the caller pipeline.

I can make the change, if you okay it.

sibanez12, 2019-04-28:
This sounds good. Please do add in this paragraph to the documentation. I can supplement with a few more details after you add this in.

Thanks :)

AnotherKamila, 2019-04-30:
Done.

It might make sense to split out the externs info from the "Wokflow Overview" wiki page, as by now it is very long, but that's for you to consider.

closed 2019-04-30 by AnotherKamila