NetFPGA/P4-NetFPGA-public #17: Issue with creating a new P4 project

nniranjhana, 2019-05-06:
Hi P4->NetFPGA community,
We have tried the switch_calc tutorial and now suppose I want to create and deploy my own P4 application from scratch in this repo. So I followed the first tip on [this](https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Getting-Started#tips).

How would I go about the compilation process for this new project?

I tried following the one for switch_calc and ran make in the $P4_PROJECT_DIR without changing the P4 program template in the new project, or the gen_testdata.py and ran into the following error:

```
fpga@phani-Precision-T5610:~/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test$ make
make -C src/ clean
make[1]: Entering directory `/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test/src'
rm -f *.sdnet *.tbl .sdnet_switch_info.dat
make[1]: Leaving directory `/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test/src'
make -C testdata/ clean
make[1]: Entering directory `/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test/testdata'
rm -f *.pcap *.txt *.pyc *.axi config_writes.* *_reg_defines.py
make[1]: Leaving directory `/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test/testdata'
rm -rf nf_sume_sdnet_ip/
rm -f 
rm -f sw/config_tables.c
make -C src/
make[1]: Entering directory `/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test/src'
p4c-sdnet -o nj_test.sdnet --sdnet_info .sdnet_switch_info.dat nj_test.p4
/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/bin/p4_px_tables.py commands.txt .sdnet_switch_info.dat
make[1]: Leaving directory `/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test/src'
make -C testdata/
make[1]: Entering directory `/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test/testdata'
./gen_testdata.py
WARNING: No route found for IPv6 destination :: (no default route?)
Traceback (most recent call last):
  File "./gen_testdata.py", line 103, in <module>
    write_pcap_files()
  File "./gen_testdata.py", line 84, in write_pcap_files
    wrpcap("src.pcap", pktsApplied)
  File "/usr/lib/python2.7/dist-packages/scapy/utils.py", line 470, in wrpcap
    PcapWriter(filename, *args, **kargs).write(pkt)
  File "/usr/lib/python2.7/dist-packages/scapy/utils.py", line 648, in write
    self._write_header(pkt)
  File "/usr/lib/python2.7/dist-packages/scapy/utils.py", line 683, in _write_header
    pkt = pkt[0]
IndexError: list index out of range
make[1]: *** [all] Error 1
make[1]: Leaving directory `/home/fpga/nj/P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/nj_test/testdata'
make: *** [frontend] Error 2
```

Also, is there a way to bypass the gen_testdata.py script checks and directly compile the P4 program? Is it recommended? As I am aware, I need to change the script according to my P4 program.

nniranjhana, 2019-05-08:
Hi @sibanez12 can you check this out? I'd like to know how to go about the compilation process for a new project, apart from the tutorials?

sibanez12, 2019-05-08:
The gen_testdata.py script assumes that you want to run at least one packet through your P4 program in simulation. For example,
```
MAC1 = "08:00:00:00:00:01"
MAC2 = "08:00:00:00:00:02"
IP1 = "10.0.0.1"
IP2 = "10.0.0.2"

pktCnt = 0

pkt = Ether(dst=MAC2, src=MAC1) / IP(dst=IP2, src=IP1)
pkt = pad_pkt(pkt, 64)
applyPkt(pkt, 'nf0', pktCnt)
pktCnt += 1
expPkt(pkt, 'nf1')
```

You can bypass the gen_testdata.py scripts by commenting out [this line](https://github.com/NetFPGA/P4-NetFPGA-live/blob/master/contrib-projects/sume-sdnet-switch/projects/switch_calc/Makefile#L61) in the Makefile. Although I'm not sure that is wise. You're going to want to test your P4 program eventually.

nniranjhana, 2019-05-08:
Ah I see, thanks a lot!!

nniranjhana, 2019-05-08:
> The gen_testdata.py script assumes that you want to run at least one packet through your P4 program in simulation.

This is why I seem to have got the list index out of range error. It makes sense now. I will append the necessary code like you've stated and not bypass the gen_testdata.py and try. I guess compilation otherwise remains the same as for the other tutorial.