NetFPGA/P4-NetFPGA-public #19: No bistream generated

mdumitru, 2019-05-14:
Doing "Assignment 1: Switch Calculator" of the [tutorials](https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Tutorial-Assignments), step 10 completes successfully, but no `.bit` files are actually generated. The folder `bitfiles` only contains `program_switch.sh` and `README`. A `find` for `*bit` files reveals that they were not created anywhere else.

I'm working on branch "vivado-2018.2", using both Vivado version 2018.2.

Here is the [vivado.log](https://github.com/NetFPGA/P4-NetFPGA-public/files/3177499/vivado.log).

sibanez12, 2019-07-01:
I see the following error message in your log file:

```
##    puts "ERROR: This script was created for Vivado <$scripts_vivado_version> and is being run in <$current_vivado_version> of Vivado. Please run the script in Vivado <$scripts_vivado_version> then open the design in Vivado <$current_vivado_version>. Upgrade the design by running \"Tools => Report => Report IP Status...\", then run write_bd_tcl to create an updated script."
```

I would suggest trying again using Vivado 2018.2 and the master branch of the P4-NetFPGA-live repo.

sibanez12, 2019-07-01:
It looks like the compilation never finished. How long did the build run for? For some reason the job was killed prematurely.

mdumitru, 2019-07-09:
The build runs for a little more than 2 hours.

I'm using lmgrd to serve a "Xilinx.lic" floating license. The lmgrd outputs these two errors several time (each line repeated 8 times):

`UNSUPPORTED: "xc7vx690t" (PORT_AT_HOST_PLUS   ) mdumitru@omicron  (No such feature exists. (-5,346))`
...
`UNSUPPORTED: "Internal_bitstream" (PORT_AT_HOST_PLUS   ) mdumitru@omicron  (No such feature exists. (-5,346))`

Unfortunately, I have no prior experience with Vivado; is this simply an issue of not having the correct correct licenses?

gsankara, 2019-09-08:
I am facing same issue during bitstream generation step where the make script getting stuck looking for impl_1. Even when I launch_runs impl_1, it never finishes.

Any issues ?

regs
Ganesh
-------------------------

Opening simple_sume_switch Implementation design

# open_run impl_1
ERROR: [Common 17-69] Command failed: Run 'impl_1' has not been launched. Unable to open
Vivado%

samersh23, 2020-01-13:
im getting an error at the same step, im getting this error:
ERROR: [DRC NSTD-1] Unspecified I/O Standard: 16 out of 62 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: pcie_7x_mgt_txn[7:0], and pcie_7x_mgt_txp[7:0].
any known issues about this error?

wangmengsail, 2020-10-17:
I should use the updated new version of P4-FPGA issued two years ago  ,which is for vivado 2018.2