NetFPGA/P4-NetFPGA-public #21: Degraded throughput with Jambo frame ( MTU 9000Bytes)

dboru, 2019-05-29:
Hi,

I have two Linux computers connected to NetFPGA-SUME board. I  configured an MTU size of 9000 Bytes for the Linux nodes and implemented a simple P4 program which does port in port out. 

I used an IPerf to test the rate between the Linux nodes and obtained literally zero throughput.  

iperf3 -c 20.0.2.4 -i 1 -t 60
Connecting to host 20.0.2.4, port 5201
[  4] local 18.0.1.4 port 41528 connected to 20.0.2.4 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec   297 KBytes  2.43 Mbits/sec    3   8.74 KBytes
[  4]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec    1   8.74 KBytes
[  4]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec    0   8.74 KBytes
[  4]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   8.74 KBytes
^C[  4]   4.00-4.93   sec  0.00 Bytes  0.00 bits/sec    0   8.74 KBytes

A tcpdump at the Linux nodes shows the following 
listening on enp101s0f1, link-type EN10MB (Ethernet), capture size 262144 bytes
15:38:15.845825 IP truncated-ip - 7502 bytes missing! (tos 0x2,ECT(0), ttl 64, id 14314, offset 0, flags [DF], proto TCP (6), length 9000)
    18.0.1.4.41524 > 20.0.2.4.5201: Flags [.], seq 2549289604:2549298552, ack 967224288, win 210, options [nop,nop,TS val 2138173070 ecr 4149489059], length 8948

Then, I added the IP checksum computation to the P4 code which has not solved the problem. 

I don't see this problem with MTU of 1500 bytes. 

What should I do to overcome this limitation? Is the MTU size fixed to a maximum of 1500 bytes for the NetFPGA-SUME board?

Thanks, 

/Dejene

AnotherKamila, 2019-06-18:
See https://github.com/NetFPGA/P4-NetFPGA-live/issues/16 -- jumbo frames are not supported, you'd have to modify the Verilog wrappers to add support for them.

dboru, 2019-06-21:
> See [NetFPGA/P4-NetFPGA-live#16](https://github.com/NetFPGA/P4-NetFPGA-live/issues/16) -- jumbo frames are not supported, you'd have to modify the Verilog wrappers to add support for them.

Thanks, Kamila for confirming. I have also tried to modify the Verilog files but no success. 

So I am closing this issue.

closed 2019-06-21 by dboru

anmomu, 2022-10-21:
> Thanks, Kamila for confirming. I have also tried to modify the Verilog files but no success.
>
> So I am closing this issue.

When modifying the wrappers and testing with `iperf` again, did you have to reload the bitstream into the FPGA because the design stopped working when sending frames of 9000 bytes?