NetFPGA/P4-NetFPGA-public #23: Working with registers with size != 32 bits

AnotherKamila, 2019-06-26:
Hello,

from reading the sources for `sume_reg_if.h`, I assume that your API only support 32-bit registers. How can I write into a register that is bigger than that (128 bits in my case)?

Thank you!

sibanez12, 2019-07-01:
While possible, I suspect this would require quite a lot of changes to both the HDL and the driver software. The easiest option for you would be to use four 32-bit registers, then each packet would read each of the registers and concatenate the bits together in the appropriate order, and the control-plane can configure each 32-bit word separately. Would that work for you?

AnotherKamila, 2019-07-08:
Hi,

I can do the 4 registers trick, but I suspect that it won't improve my already existing timing problem. But if you think that lots of changes would be require, I suppose I will try to go with that option.

Thank you for the info! Closing this then.

closed 2019-07-08 by AnotherKamila