NetFPGA/P4-NetFPGA-public #24: Step 4: Link to Xilinx SDNet v2018.2 is incorrect

harshgondaliya, 2019-06-26:
The hyperlink in Step 4 is not correct. It leads to a page where only Product Guides and documents for Xilinx SDNet v2018.2 are there. The zip file package download is not available anywhere.

I also checked through Xilinx website but I could not find Xilinx SDNet v2018.2 download. May I please get link to the download?

harshgondaliya, 2019-06-26:
https://forums.xilinx.com/t5/Installation-and-Licensing/Download-link-for-latest-version-of-Xilinx-SDNet/td-p/980543

Some relevant discussion on Xilinx community forum in this regard

dboru, 2019-06-26:
I think it has been migrated to the following 
https://sanet.st/blogs/downloaddownload/xilinx_sdnet.2748007.html