NetFPGA/P4-NetFPGA-public #27: sim_switch_ctrlWrites : No such file or directory

harshgondaliya, 2019-06-30:
`sim_switch_ctrlWrites ` directory has been provided for `switch_calc` tutorial project.

But, for all other projects i.e. INT, learning_switch, tcp_monitor there is no` sim_switch_ctrlWrites`  directory at path` $NF_DESIGN_DIR/test/`

Step 8-9 of compilation process requires us to run `sim_switch_ctrlWrites `simulation. Does it mean that `sim_switch_ctrlWrites `simulation need not be run for INT, learning_switch and tcp_monitor project?
What is role of this simulation in compilation process?

sibanez12, 2019-07-01:
That's right, the switch calc project is the only one with the `sim_switch_ctrlWrites` simulation. It is just intended to be an example of how you can read/write P4 registers during a simulation. Feel free to use it as a reference.

harshgondaliya, 2019-07-02:
OK, got it. Thanks.

closed 2019-07-02 by harshgondaliya