NetFPGA/P4-NetFPGA-public #37: step 10: Compiling the bitstream

samersh23, 2019-12-31:
Hey,
we tried first to run using vivado 2019.2 and a different error appeared which asked us to use vivado 2016.4, after downloading 2016.4 and ran step 10 again we get this error:
ERROR: [BD 41-171] The modes of the interface pins 'cfg_interrupt'(Slave) and 'pcie3_cfg_interrupt'(Slave) are incompatible. They cannot be connected.
ERROR: [BD 5-3] Error: running connect_bd_intf_net.
ERROR: [Common 17-39] 'connect_bd_intf_net' failed due to earlier errors.

has anyone encountered this issue?

salvatorg, 2020-01-18:
Quote from the wiki.

> Important: This project uses Vivado 2018.2. You will likely run into issues if you do not use this version of Vivado

https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Getting-Started#steps-for-initial-setup

samersh23, 2020-01-22:
there are some scripts that needs to be run on vivado 2016.4 (while compiling the bitstream), we are having some issues when to run vivado 2018.2 and when to run vivado 2016.4 and how to adapt the results when switching between the two versions.

salvatorg, 2020-01-28:
Can you point out the scripts that need Vivado v2016.4 please.

samersh23, 2020-02-04:
P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/learning_switch/simple_sume_switch/hw/tcl/control_sub.tcl
P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/learning_switch/simple_sume_switch/hw/tcl/control_sub_sim.tcl

salvatorg, 2020-02-04:
> P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/learning_switch/simple_sume_switch/hw/tcl/control_sub.tcl
> P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/learning_switch/simple_sume_switch/hw/tcl/control_sub_sim.tcl

These scripts will run only with Vivado version 2018.2 otherwise they will throw an error.
What error are you getting from these scripts?

> a different error appeared which asked us to use vivado 2016.4
Can you copy paste/reproduce the message you are getting?

samersh23, 2020-02-04:
thanks for the quick response, when trying to create the bitstream, we get from these scripts error specifying that we are running with version 2018.2 but these scripts are written for Vivado 2016.4.

I tried to avoid the error by changing the wanted Vivado version for this scripts to 2018.2, but then I got an error for invalid ip:pcie3 version (the error is coming from the script control_sub.tcl), and from searching the internet I found out that its a known issue and it's happening because I am running the scripts with a newer Vivado version which expect a newer ip:pcie3 version.....

salvatorg, 2020-02-17:
The P4-NetFPGA repository runs with Vivado v2018.2 . The NetFPGA-SUME repository runs with Vivado v2016.4 .
Make sure before running any script in the P4-NetFPGA directory that you have correctly source the settings.sh under that directory and the correct version of Vivado. Check the version with : `vivado -version`

samersh23, 2020-02-17:
I think we ran all the scripts either with 2018 or with 2016. I will restart all the process and run each repository steps with the correct vivado version. Thanks a lot

species-gh, 2020-03-23:
@samersh23, did you eventually find a solution? I'm running into exactly the same error messages with Vivado 2018.2, trying to create the hardware project. 2016.4 (as expected) fails even much earlier:
```
ERROR: This script was created for Vivado <2018.2> and is being run in <2016.4> of Vivado.
Please run the script in Vivado <2018.2> then open the design in Vivado <2016.4>.
Upgrade the design by running "Tools => Report => Report IP Status...", then run write_bd_tcl to create an updated script.
```
```
ERROR: [Coretcl 2-1133] Requested IP 'xilinx.com:ip:fifo_generator:13.2' cannot be created.
The latest available version in the catalog is 'xilinx.com:ip:fifo_generator:13.1'.
If you do not wish to select a specific version please omit the version field from the command arguments, or use a wildcard in the VLNV.

    while executing
"source ./create_ip/nf_10ge_interface.tcl"
    (file "tcl/simple_sume_switch.tcl" line 97)

```

samersh23, 2020-03-23:
I tried to do what @salvatorg said above, and as you said i got the same error messages regarding the 2018.2 and 2016.4 version although I did all the **steps with the wanted versions**.
and as you said it happens some steps before the bitstream step.
I'm not sure how to continue my work.

samersh23, 2020-03-23:
It happens in step 9 in the  **SUME simulation**

salvatorg, 2020-03-23:
> @samersh23, did you eventually find a solution? I'm running into exactly the same error messages with Vivado 2018.2, trying to create the hardware project. 2016.4 (as expected) fails even much [...]

Thats because you are running Vivado v2016.4 and IP 'xilinx.com:ip:fifo_generator:13.2' doesnt exist in that version of vivado. The Vivado v2018.2 has the particular version of ip.
If you use P4-NetFPGA repo, be sure that you use **only** Vivado v2018.2

salvatorg, 2020-03-23:
> I tried to do what @salvatorg said above, and as you said i got the same error messages regarding the 2018.2 and 2016.4 version although I did all the **steps with the wanted versions**.
> and as you said it happens some steps before the bitstream step.
> I'm not sure how to continue my work.

I assume you did all the step using Vivado 2018.2, correct?
What error are you getting in in step 9 in the SUME simulation?

samersh23, 2020-03-23:
I ran The P4-NetFPGA repository with Vivado v2018.2, and the NetFPGA-SUME repository with Vivado v2016.4 .
and I'm getting the same error written above 

> ``
> ERROR: This script was created for Vivado <2018.2> and is being run in <2016.4> of Vivado.
> Please run the script in Vivado <2018.2> then open the design in Vivado <2016.4>.
> Upgrade the design by [...]

species-gh, 2020-03-23:
For me it seems to work using 2018.2 now, after I started all over again with a freshly cloned repo (synth still running; good sign).
Seems like the the [errors above](#issue-544200794) emerged from starting with 2018.**3** and then switching to 2018**.2**, when other steps failed in **.3**.
@salvatorg, make sure to also do step 8 in [Getting Started](https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Getting-Started) with 2018.2 and then follow the steps from the tutorial assignments.

samersh23, 2020-03-23:
did you run  The P4-NetFPGA repository with Vivado v2018.2, and the NetFPGA-SUME repository with Vivado v2016.4 ?
or how did you manage to solve it?

species-gh, 2020-03-23:
I'm only working with the P4-NetFPGA-live repo, not with the NetFPGA-SUME repo

species-gh, 2020-03-23:
No, I didn't have to touch any scripts. Just start from a clean repo and to everything with 2018.2

samersh23, 2020-03-23:
in the getting started link you attached before, in the first step it is written : Go to the NetFPGA SUME System Setup wiki page and follow the instructions. Make sure to install all of the required dependencies.

didnt you do the NetFPGA SUME System Setup?

species-gh, 2020-03-23:
> @species-gh in the getting started link you attached before, in the first step it is written : Go to the NetFPGA SUME System Setup wiki page and follow the instructions. Make sure to install all of the required dependencies.
> 
> didn't you do the NetFPGA SUME System Setup?

Not entirely, as I'm only developing on my local machine, but the board is placed in a different machine. All I needed from there is the list of dependencies to install. Also don't forget to install the dependencies from _Getting Started_, especially `libc6-dev-i386`.

samersh23, 2020-03-23:
and are using the board as standalone or connected to the motherboard?
and can you explain more on why you only installed the list of dependencies from the Getting Started? why not the rest?

species-gh, 2020-03-23:
The board is connected to the motherboard of a server. 
Regarding your second question: I was referring to the [_Reference Operating System Setup Guide_](https://github.com/NetFPGA/NetFPGA-SUME-public/wiki/Reference-Operating-System-Setup-Guide), when I was talking about the list of dependencies. That page contains some instructions, that don't seem to be necessary unless actual errors occur, that can be tracked down to not following those hints (e.g. enabling root login and messing with avahi and the network manager). Also, some of the instructions and dependencies clearly are only necessary for systems connected to the board (e.g. `minicom` and `libusb-dev`). So I tried to only install the packages I really need and avoided messing with my OS (I'm on my private machine right now due to the Covid-19 outbreak in Germany).
Regarding the _Getting Started_ guide itself, I did follow most of it. I only skipped step 9, since it's the driver installation for communication with the board via PCIe (of course I had to do this on the server, but not on my current development machine).

samersh23, 2020-03-23:
I think my mistake was doing all the steps in every tutorial/setup guide.
I will try again with a clean new repo.
thank you very much for the help :)

closed 2020-04-01 by samersh23