NetFPGA/P4-NetFPGA-public #41: Prioritizing overlapping table entries in ternary match

harshgondaliya, 2020-02-03:
Using P4-NetFPGA Workflow, I want to learn how priority can be set for overlapping entries in the ternary match.

Ex:
_commands.txt file_
```
table_tcam_add_entry lookup_ip_dstAddr 0x1 ipv4_forward 1.1.1.0/255.255.255.0 => 03:03:03:03:03:03 0b00000001

table_tcam_add_entry lookup_ip_dstAddr 0x2 ipv4_forward 1.1.1.0/255.255.255.0 => 04:04:04:04:04:04 0b00000100
```

How do I prioritize between the above two entries?

Any explanation in this regard will be really helpful.

guimvmatos, 2020-11-09:
Hi @harshgondaliya 

I think maybe you’ve already solved your problem, but you can prioritize by address. Think of these addresses as a rule priority. Minors have priority.

harshgondaliya, 2020-11-09:
Oh yes, got it. Actually, I had not got a solution so far. Thanks for your reply.

closed 2020-11-09 by harshgondaliya