NetFPGA/P4-NetFPGA-public #46: SUME simulation fails because of pkt.tuser_sport

species-gh, 2020-04-01:
Hi,
I'm currently trying to setup the SUME simulation as described in the [switch-calc tutorial in step 9](/NetFPGA/P4-NetFPGA-public/wiki/Tutorial-Assignments#what-to-do).
Running the simulation with
```
$ ./tools/scripts/nf_test.py sim --major switch --minor default --gui
```
Vivado opens as intended, but aborts the simulation. The console tells me, that `run.py` threw an `AttributeError` (username and exact path replaced by '\<username\>'):
```
loading libsume..
About to start the test
/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/testdata/nf1_applied.pcap  not found
/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/testdata/nf2_applied.pcap  not found
/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/testdata/nf3_applied.pcap  not found
/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/testdata/nf1_expected.pcap  not found
/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/testdata/nf2_expected.pcap  not found
/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/testdata/nf3_expected.pcap  not found
scheduling pkts ...
Traceback (most recent call last):
  File "/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/simple_sume_switch/test/sim_switch_default/run.py", line 92, in <module>
    schedule_pkts(nf0_applied, 'nf0')
  File "/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/simple_sume_switch/test/sim_switch_default/run.py", line 63, in schedule_pkts
    pkt.tuser_sport = nf_port_map[iface]
  File "/usr/lib/python2.7/dist-packages/scapy/packet.py", line 221, in __setattr__
    return object.__setattr__(self, attr, val)
AttributeError: 'Ether' object has no attribute 'tuser_sport'
    while executing
"exec python $::env(NF_DESIGN_DIR)/test/${test_name}/run.py"
    invoked from within
"set output [exec python $::env(NF_DESIGN_DIR)/test/${test_name}/run.py]"
    (file "/home/<username>/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/switch_calc/simple_sume_switch/hw/tcl/simple_sume_switch_sim.tcl" line 177)
```

Origin of the error is the method `schedule_pkts()` in `run.py`, where the attribute `tuser_sport` is added dynamically to a Scapy packet instance:
```
def schedule_pkts(pkt_list, iface):
    print 'scheduling pkts ...'
    for pkt in pkt_list:
        pkt.time = baseTime + delta*pkt.time
        pkt.tuser_sport = nf_port_map[iface]
    print 'done scheduling pkts ...'
```
I could find even more attempts to add this attribute in other files, that will fail for sure, too.

After a lot of web search I could trace down the cause of the problem to be a [change in Scapy four years ago](/secdev/scapy/commit/09e12802fc2694f5de2870b600c2abe732705366), that introduced slots to most base classes, which prevents the dynamic addition of attributes.

Odd, that seemingly no one else here had stumbled upon this yet. :sweat_smile: 

#### Additional information
**OS:** Ubuntu 18.04
**Python:** 2.7.17
**Scapy:** 2.3.3

salvatorg, 2020-04-02:
Hi @species-gh,
Thnx for pointing this out! I personally havent seen that error apparently because I have scapy version 2.2.0 in my system.

ralfkundel, 2020-04-02:
Hi,
i just tested in on multiple versions:
2.2.0 : working
2.3.3 : not working (apt install python-scapy on Ubuntu 18.04)
3.0.0 : working

regards,
Ralf

species-gh, 2020-04-02:
@ralfkundel found that adding attributes to packets works again in newer Scapy versions. So I switched to the current version available from the Scapy repo (2.4.3) and now the errors are gone. No need for a fix in the scripts :+1:

closed 2020-04-02 by species-gh