NetFPGA/P4-NetFPGA-public #47: Recirculated packets are not emitted by the SDNet module

species-gh, 2020-04-11:
Hi,
for this issue I first have to explain my setup: I've created a loop-back path from the axis output of the SDNet IP core (simple sume switch) to the input arbiter. For this I added a sixth input to the arbiter and reused the deprecated `drop` field in the sume metadata to control, where packets are sent to (output queues or loop) and to indicate to the simple sume switch, if a packet was recirculated through the loop. Furthermore, I increased the data bus width to 512 bits.

Now, if I use the SDNet simulation, everything is working just right. I can read and manipulate the loop-control metadata and also simulate packets, that already passed the loop using `gen_testdata.py`. These tests always pass, packets are processed and emitted just as expected, no matter if they are "fresh from the input" or already marked as recirculated.

If I go a step further and run the SUME simulation (using a single packet), I can observe the packet being processed and looped back correctly. However, the simple sume switch module does not emit anything after receiving the recirculated packet again.
At this point I'm stuck, as I cannot look into what's happening inside of the simple sume switch. Does anyone have an idea, what could be the problem here? If anyone wants to reproduce this, I could provide my current code.

species-gh, 2020-04-14:
Some new insights:
- Setting all my metadata inside of the former `drop` field to 0 does not change anything. Therefore I can rule out this as the cause.
- Not setting the dst-port in P4, so the recirculated packet does not have the dst-port already set when it re-enters the simple sume switch, also does not change anything (I'm clutching at every straw now...)
- I noticed, that actually I do can look into the simple sume switch module and I found, that the recirculated packet doesn't even reach the `TopParser`. Unfortunately the internal structure of the module is unclear to me, I was just guessing, that the `TopParser` must be one of the first internal modules. At least this tells me, that the problem must be somewhere at the input side of the simple sume switch.

Looking at the axis interface everything seems alright. So this is what the initial packet from interface `nf0` looks like:
[image]
Value of `s_axis_tdata` is `00000000000000000000000000000000000000000000000000000000000000000000000000000000000400000000010000001212081111111108082222222208`, corresponding to the scapy packet `Ether(dst="08:22:22:22:22:08", src="08:11:11:11:11:08") / Calc(op1=1, opCode=ADD_OP, op2=4, result=0)`

And this is what the packet looks like on the output of the simple sume switch (yellow marker) and after recirculation on the input side again (blue marker):
[image]
Value of `m_axis_tdata` (`s_axis_tdata` respectively) is `00000000000000000000000000000000000000000000000000000000000000000000000000050000000400000000010000001212082222222208081111111108`, corresponding to `Ether(dst="08:11:11:11:11:08", src="08:22:22:22:22:08") / Calc(op1=1, opCode=ADD_OP, op2=4, result=5)`, just as expected

species-gh, 2020-04-14:
Some more updates on this:
- Sending packets longer than 64 bytes makes the SDNet module scramble up the output completely. In fact, the output inserts the beginning of the packet every 4 clock cycles, which is obviously nonsense.
- I changed the data bus width back to 256 bit and now it's working, including my metadata for controlling the recirculation.

So my conclusion is that the hdl created by SDNet for the sume simulation does not work with 512 bit data bus width, even though the SDNet compiler runs without errors using `-busWidth 512`. I'll change the topic of this issue accordingly

salvatorg, 2020-04-14:
> Some more updates on this:
>
> * Sending packets longer than 64 bytes makes the SDNet module scramble up the output completely. In fact, the output inserts the beginning of the packet every 4 clock [...]

Thnx for all the insights @species-gh, are very useful and its great that you found the issue.

> I changed the data bus width back to 256 bit and now it's working, including my metadata for controlling the recirculation.

I would suspect that was something going wrong with the bus width because the SUME pipeline uses 256b. Changing the parameter in the SDNet compiler doesnt mean that you also change the width of the pipeline.

>Sending packets longer than 64 bytes makes the SDNet module scramble up the output completely. In fact, the output inserts the beginning of the packet every 4 clock cycles, which is obviously nonsense.

You mean that every packet is processed with a 4 cc gap?

species-gh, 2020-04-15:
Thanks for your reply, @salvatorg !

> I would suspect that was something going wrong with the bus width because the SUME pipeline uses 256b. Changing the parameter in the SDNet compiler doesnt mean that you also change the width of the pipeline.

What exactly do you mean with 'the SUME pipeline'? An internal part of the SDNet module or the whole datapath? In fact, I did increase the bus width of the complete datapath, starting at the 10G interface wrapper modules, over the input arbiter and SDNet module to the output queues, of course including all the connections in the `top`/`top_sim` module and the `nf_datapath` module and the axis test data creation in python. Took me a while to find all places, where 256b were hard coded. :wink:

> You mean that every packet is processed with a 4 cc gap?

What I did was increasing the size of an Ether/Calc packet with about 1kB of (nonsense) raw data (repeating `0xCAFE`) to find out, if it was a problem related to small packets. In the following screenshot you can see the the packet starting with the Ether and Calc headers in the first cycle and then the data bus stays at `...cafecafe...` until the complete packet is sent (This is the axis input to the SDNet module. Signals are from top to bottom: `tdata`, `tkeep`, `tvalid`, `tready`, `tlast`).
[image]
On the output side I would expect to see a similar picture, but what I found was the Ether and Calc headers being inserted into the data bus every four cycles until the whole length of the packet was transmitted:
[image]

species-gh, 2020-06-03:
Some good news on this: I've just found the source of the weird behavior with long packets and it has nothing to do with the databus width. In fact, I did also reproduce it on a clean repo with default 256bit databus. The problem is hidden in the P4 code of the switch_calc project, that I'm using as a base for development.
Lines 97 and 201 state:
```
@Xilinx_MaxPacketRegion(1024)
```
For all the other example projects this value is set much higher (16384). The "P4-SDNet Translator User Guide" (UG1252) describes this value like this: _"The maximum packet depth (in bits).Default: 12,144"_ 

**tl;dr:** Using `@Xilinx_MaxPacketRegion(16384)` fixes the issue with long packets.

With some more time I'll try out, If this also fixes the initial issue with disappearing packets, when the bus width is 512 bit.

species-gh, 2020-06-17:
This indeed also fixes the initial weird behavior with long packets and 512 bit wide axis_tdata,
so I'm closing this issue.

closed 2020-06-17 by species-gh

species-gh, 2020-06-17:
Sorry, I was a little too fast with my prior comment. I did another simulation, this time with 64byte packets as in my initial setup, when I opened this issue. The simple_sume_switch still "swallows" some packets. I'm sending 4 packets, each should be recirculated once, which makes a total of 8 transmissions, that should be visible on the simple_sume_switch's axis output. However, there are only 3 packets/transmissions visible on `m_axis_opl`, meaning that some packets just vanished.

So, to put this in a nutshell: The errors still persist and are most probably caused by an issue in the module compiled by SDNet. I'd keep this issue closed, though, because this is nothing, that can be fixed in the code of P4-NetFPGA, but rather is an issue of SDNet.

species-gh, 2020-06-17:
Once again I have to correct myself, sorry for that. But I have some really good news: I've just found the source of the problem and fixed it!
I couldn't really believe, that Xilinx screwed up the SDNet compiler for 512 bit, while the user guide actually states that up to 1024 bit are supported. So I had a look into the simple sume switch wrapper module. Initially I had in mind, that it does only pass the axis signals into the SDNet module, so I didn't look there. But now I indeed found the issue in `sume_to_sdnet.v`:
```
always @(*) begin
   // defaults
   state_next = state;
   SDNet_tuple_VALID = 0;

   case(state)
     /* wait to complete first cycle of packet */
     FIRST: begin
         if (SUME_axis_tvalid & SUME_axis_tready) begin
             SDNet_tuple_VALID = 1;
             state_next = WAIT;
         end
     end

     /* wait until last cycle of packet */
     WAIT: begin
         if (SUME_axis_tvalid & SUME_axis_tlast & SUME_axis_tready) begin
             state_next = FIRST;
         end
     end // case: WAIT

   endcase // case(state)
end // always @ (*)
```
Increasing the databus width to 512bit means, that 64byte packets only take one clock-cycle. Such a packet leaves the upper state machine in the `WAIT` state without returning to `FIRST`. On arrival of the next packet, `SDNet_tuple_VALID` will be 0 and the SDNet module will not sample the data, making the packet just vanish. However, it also brings the state machine back to `FIRST` state, so the next packet will be processed again. Testing for `tlast` in the `FIRST` state fixes this. It's as simple as that:
```
     FIRST: begin
         if (SUME_axis_tvalid & SUME_axis_tready) begin
             SDNet_tuple_VALID = 1;
             if (SUME_axis_tlast)
                 state_next = FIRST;
             else
                 state_next = WAIT;
         end
     end
```
I did check this in the sume simulation and it indeed fixes the issue. So now this may stay closed for real. I can create a pull request later, if you like ;)