NetFPGA/P4-NetFPGA-public #48: No Such Device issue

samersh23, 2020-05-12:
we are working on the project TCP Monitor from the tutorial assignments page. we successfully installed the sume_riffa driver, when running lsmod (after running modprobe sume_riffa) we can see the sume_driver in the list. but unfortunately, we can't see the nf0 nf1 nf2 nf3 interfaces.
On the FPGA side, the card is connected to the PC with a USB, and to 2 ports of the FPGA we had 10G NIC connection (just as described at the beginning of the tutorials page - https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Tutorial-Assignments).

we are running on Linux 18.04 and kernel 4.15, we saw that it's recommended to use kernel 4.10.0.42. does anyone get this problem? does anyone know if changing the kernel would solve it?
thanks in advance.

species-gh, 2020-05-12:
You need to reboot the machine after programming the FPGA. Did you do that?

samersh23, 2020-05-12:
Yes, and also did the driver installation as mentioned in the assignment in step 13. but I'm still not getting the nf0 nf1 nf2 nf3 interfaces...
I'm suspecting the kernel version is the issue, but I'm not sure

ralfkundel, 2020-05-12:
Hi,

do you see the FPGA in lspci and is the FPGA connected to the PCIe
connector?

best,
Ralf

-- 
________________________________________________
Ralf Kundel, M.Sc.

Technical University of Darmstadt
Dept. of Electrical Engineering and Information Technology Multimedia Communications Lab (KOM)
Adaptive Communication Systems Group (ACS)

Postal address: TU Darmstadt, FB 18, KOM, Rundeturmstr. 10, 64283 Darmstadt, Germany
Location: S3|20 - 216
Tel: +49 6151 16-20855
Email: Ralf.Kundel@kom.tu-darmstadt.de
Web: http://www.kom.tu-darmstadt.de

samersh23, 2020-05-12:
the FPGA is in standAlone connection, it is connected to the PC by a USB.
I'm not sure if this answers your question, but when connecting/disconnecting the USB from the PC we can see these actions in the dmesg... we can see that "NETFPGA Sume driver was connected"...

ralfkundel, 2020-05-12:
I never tried this and I'm not sure if the kernel module is able to
operate over USB/JTAG.

-- 
________________________________________________
Ralf Kundel, M.Sc.

Technical University of Darmstadt
Dept. of Electrical Engineering and Information Technology Multimedia Communications Lab (KOM)
Adaptive Communication Systems Group (ACS)

Postal address: TU Darmstadt, FB 18, KOM, Rundeturmstr. 10, 64283 Darmstadt, Germany
Location: S3|20 - 216
Tel: +49 6151 16-20855
Email: Ralf.Kundel@kom.tu-darmstadt.de
Web: http://www.kom.tu-darmstadt.de

salvatorg, 2020-05-12:
> […](#)
> ________________________________________________ Ralf Kundel, M.Sc. Technical University of Darmstadt Dept. of Electrical Engineering and Information Technology Multimedia Communications Lab [...]

Exactly as @ralfkundel said. The driver uses the PCIe end points to communicate with the host/board. There is no implementation over the USB. You ll have to plug the board in the PCIe slot to have the register access.

samersh23, 2020-05-13:
Thanks for the help. We connected the NetFPGA to the PCIe, but unfortunately, we couldn't see the NetFPGA in the lspci. It appears that we don't have PCIe GEN3. Until we find such PC, we wonder what still we can do in “standalone” mode. From what we understood, we successfully burned the simple-switch bitstream file. We can see that we get link up when the NetFPGA ports are connected to an external device. Is there a way to manage the NetFPGA via the USB serial port – i.e read or write P4 match-action tables or read registers ?