NetFPGA/P4-NetFPGA-public #51: KeyError: 'thresh' when running run.py for fred_basic

harshgondaliya, 2020-09-23:
I am running the SUME SIMULATION for fred_basic example program by executing the following commands:

`cd $NF_DESIGN_DIR/test/sim_switch_2flows && make
`
`cd $SUME_FOLDER && ./tools/scripts/nf_test.py sim --major switch --minor 2flows 
`

I face the following error from run.py script.
```
WARNING: [HDL 9-2994] overwriting previous definition of module fallthrough_small_fifo [/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/hw/project/simple_sume_switch.srcs/sources_1/ip/sume_event_switch_ip/sume_event_switch_ip/SimpleSumeSwitch/qsize_reg_multi_raws_0_t.HDL/fallthrough_small_fifo.v:46]
WARNING: [HDL 9-2994] overwriting previous definition of module small_fifo [/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/hw/project/simple_sume_switch.srcs/sources_1/ip/sume_event_switch_ip/sume_event_switch_ip/SimpleSumeSwitch/qsize_reg_multi_raws_0_t.HDL/small_fifo.v:44]
update_compile_order: Time (s): cpu = 00:00:15 ; elapsed = 00:00:06 . Memory (MB): peak = 2011.980 ; gain = 8.004 ; free physical = 26807 ; free virtual = 32196
loading libsume..
About to start the test
Traceback (most recent call last):
  File "/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/sim_switch_2flows/run.py", line 97, in <module>
    nftest_regwrite(p4_externs['thresh']['base_addr'], THRESH)
KeyError: 'thresh'
    while executing
"exec python $::env(NF_DESIGN_DIR)/test/${test_name}/run.py"
    invoked from within
"set output [exec python $::env(NF_DESIGN_DIR)/test/${test_name}/run.py]"
    (file "/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/hw/tcl/simple_sume_switch_sim.tcl" line 164)
INFO: [Common 17-206] Exiting Vivado at Wed Sep 23 12:45:36 2020...
Makefile:120: recipe for target 'sim' failed
make: *** [sim] Error 1
make: Leaving directory '/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test'
512
=== Work directory is /tmp/root/test/simple_sume_switch
=== Setting up test in /tmp/root/test/simple_sume_switch/sim_switch_2flows
cp: cannot stat '/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/nf_interface_0_log.axi': No such file or directory
cp: cannot stat '/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/nf_interface_1_log.axi': No such file or directory
cp: cannot stat '/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/nf_interface_2_log.axi': No such file or directory
cp: cannot stat '/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/nf_interface_3_log.axi': No such file or directory
cp: cannot stat '/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/dma_0_log.axi': No such file or directory
cp: cannot stat '/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/reg_stim.log': No such file or directory
=== Running test /tmp/root/test/simple_sume_switch/sim_switch_2flows ... using cmd ['/opt/Xilinx/harsh-P4-NetFPGA/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/sim_switch_2flows/run.py', '--sim', 'xsim']
root@minion1:/opt/Xilinx/harsh-P4-NetFPGA# 
```

It seems like some previous step in the workflow has not worked properly. Any suggestions on debugging this?

rst0git, 2020-09-27:
Hi @harshgondaliya, could you please try running the `fred_basic` simulation with the following steps:
1. Set the variable `P4_PROJECT_NAME` in `tools/settings.sh` to the value of `fred_basic`.
2. Run the following script:

```bash
source tools/settings.sh

# Build Vivado core IP modules
cd $SUME_FOLDER/lib/hw/xilinx/cores/tcam_v1_1_0/ && make update && make
cd $SUME_FOLDER/lib/hw/xilinx/cores/cam_v1_1_0/ && make update && make
cd $SUME_SDNET/sw/sume && make
cd $SUME_FOLDER && make

# Build sume_riffa driver
make -C $DRIVER_FOLDER all
make -C $DRIVER_FOLDER install
modprobe sume_riffa

# Generate verilog code and API/CLI tools
make -C $P4_PROJECT_DIR

# Run SDNet simulation
cd $P4_PROJECT_DIR/nf_sume_sdnet_ip/SimpleSumeSwitch
./vivado_sim.bash

# Generate the scripts for NetFPGA SUME simulation
make -C $P4_PROJECT_DIR config_writes
make -C $P4_PROJECT_DIR install_sdnet
make -C $NF_DESIGN_DIR/test/sim_switch_default

# Run NetFPGA simulation
cd $SUME_FOLDER
./tools/scripts/nf_test.py sim --major switch --minor default
```

harshgondaliya, 2020-09-29:
Hi @rst0git,

Thanks for the detailed steps!
The SUME simulation set by `run.py` script in `$NF_DESIGN_DIR/test/sim_switch_default` directory works perfectly fine for me.

But, the SUME simulation set by `run.py` script in `$NF_DESIGN_DIR/test/sim_switch_2flows` directory gives me the `KeyError` that I described in my first post.

I think there is some error in [line#97](https://github.com/NetFPGA/P4-NetFPGA-live/blob/master/contrib-projects/sume-sdnet-switch/projects/fred_basic/simple_sume_switch/test/sim_switch_2flows/run.py#L97) of `$NF_DESIGN_DIR/test/sim_switch_2flows/run.py` script. 
`nftest_regwrite(p4_externs['thresh']['base_addr'], THRESH)
`

`fred_basic.p4` doesn't have any extern/register named `'thresh'`, still the above code is trying to fetch value for `'thresh'` extern.

rst0git, 2020-09-29:
> `fred_basic.p4` doesn't have any extern/register named 'thresh', still the above code is trying to fetch value for 'thresh' extern.

@harshgondaliya you are right, `thresh_reg_rw` appears to have been replaced with a match-action table in commit [`aaace15`](https://github.com/NetFPGA/P4-NetFPGA-live/commit/aaace1568c64243b38dba1a6a0e55dbbd044582c#diff-0720176662bd0b28ec3a1745c3cc41c1L80).

harshgondaliya, 2020-09-30:
Oh, I see. Thanks for pointing out!