NetFPGA/P4-NetFPGA-public #52: Cannot see the network port information after NetFPGA SUME driver is loaded

zhangzicheng0428, 2020-10-16:
Hello!

I now use the NetFPGA SUME development board for some tests. The experiment is completed according to the steps in the test document (such as screenshots). After loading the driver during the test, the network interface information cannot be viewed in the system (Go to step 7). The details are as follows:
[image]

1.system message：

 

CPU:Intel(R) Core(TM) i5-7400 CPU @ 3.00GHz

 

Motherboard model：Gigabyte B150M-VP

 

Operating system: Ubuntu 16.4

Kernel version: 4.15.0-120-generic

2.Test operation steps：

a.On the xsct console, use connect and then fpga -f reference_switch_lite.bit to program the FPGA with the bitfile and exit to close xsct console. Reboot the machine.

image.png

b.Check that the bitfile file is successfully burned

image.png.b1a4f32263f22c904346a006dc51a644.png

c.Build the driver for the NetFPGA SUME board.

image.thumb.png.e9ee248598a262788a5abdd2386ecd68.png

d.View network interface information in the system

image.png.0902a1a573cfc3fae11e2eeedf0e26fb.png

 

 

In the fourth step "c", you can see that there are four more interfaces after loading the driver, but these four interfaces are not the "nfx" interfaces mentioned in step 7, and cannot be viewed by using ethtool Find the interface information, and cannot UP and configure the IP address?

zhangzicheng0428, 2020-10-16:
> Hello, if you could follow instructions in this wiki page -> [ Network Setup for Ubuntu 2020.4 LTS](https://github.com/NetFPGA/NetFPGA-SUME-public/wiki/Network-Setup-for-Ubuntu-2020.4-LTS) your issue will be resolved

Thanks! Hilx 

The system I am debugging is ubuntu 16.04, the version mentioned on the connection is version 20.4, and netplan cannot be found in the etc directory on 16.04.

salvatorg, 2020-10-17:
Hi @zhangzicheng0428 ,

Have you tried [this](https://github.com/NetFPGA/NetFPGA-SUME-public/wiki/Reference-Operating-System-Setup-Guide#network-interface-names-ubuntu-1604)

closed 2021-11-05 by joygetsit