NetFPGA/P4-NetFPGA-public #55: ERROR: tuple mismatch for packet 1 (simulation with ./vivado_sim.bash .)

guimvmatos, 2020-11-05:
Hi everybody, 

I'm trying to implement my bmv2 [project](https://github.com/guimvmatos/srv6Project/blob/master/src/srv6Project.p4) on P4-NetFPGA, but I'm having some problems that I don't understand, so I decided to start with something easier, like a simple mac swap. But I'm still in trouble. All my code is [here](https://github.com/guimvmatos/srv6Project/tree/P4clean). You can see the [P4 code](https://github.com/guimvmatos/srv6Project/blob/P4clean/src/srv6Project.p4), [commands.txt](https://github.com/guimvmatos/srv6Project/blob/P4clean/src/commands.txt) and [gen_testdata.py](https://github.com/guimvmatos/srv6Project/blob/P4clean/testdata/gen_testdata.py)

after running a simulation (./vivado_sim.bash .) I am getting the error below:
```
[4566506]  INFO: finished packet stimulus file
[5014660] ERROR: tuple mismatch for packet 1
```
What am I doing wrong?
[image]

salvatorg, 2020-11-05:
Maybe wrong output interface. Double check where you send out the packet in your p4 code, either in the testdata whether its correct the output interface that you want to see the packet coming out.

guimvmatos, 2020-11-05:
> 
> 
> Maybe wrong output interface. Double check where you send out the packet in your p4 code, either in the testdata whether its correct the output interface that you want to see the packet coming out.

in fact, the expected port in gen_testdata.py was not correct. Now it's working. I'll try to generate the bitstream to test with real hardware.
Thank you @salvatorg

closed 2020-11-09 by guimvmatos