NetFPGA/P4-NetFPGA-public #56: ERROR: not enough fields provided to complete _hexify()

guimvmatos, 2020-11-09:
Hi, 

I'm trying to add an entry on a ternary table, but i'm getting this error: `ERROR: not enough fields provided to complete _hexify()`

But, I think I'm doing everything right. Can anybody tell me what's wrong?

I'm using `python sw/CLI/P4_SWITCH_CLI.py` to add entries. I've tryed with the lines below:

```
table_tcam_add_entry teid_exact 0x1 srv6_t_insert_2 0xfffffffff/0xfffffffff => 0xfc000000000000000000000000000001 0xfc000000000000000000000000000002

table_tcam_add_entry teid_exact 0x1 srv6_t_insert_2 0b00000000000000000000000000100000/0b11111111111111111111111111111111 => 0xfc000000000000000000000000000005 0xfc000000000000000000000000000006

table_tcam_add_entry teid_exact 0x1 srv6_t_insert_2 0x32/0x32 => 0xfc000000000000000000000000000005 0xfc000000000000000000000000000006

table_tcam_add_entry teid_exact 0x1 srv6_t_insert_2 0x0020/0x0020 => 0xfc000000000000000000000000000001 0xfc000000000000000000000000000002

table_tcam_add_entry teid_exact 1 srv6_t_insert_2 32/32 => 0xfc000000000000000000000000000001 0xfc000000000000000000000000000002
```

This is my table:

```
    table teid_exact{
        key = {
            hdr.gtp.teid: ternary;
     }
        actions = {
            srv6_t_insert_2;
            srv6_t_insert_3;
            NoAction;
        }
        size = 64;
        default_action = NoAction;
    }
```
And my action:

```
    action srv6_t_insert_2(ip6Addr_t s1, ip6Addr_t s2){
        hdr.ipv6_outer.payload_len = hdr.ipv6_outer.payload_len + 40;
        hdr.srv6_sid1.setValid();
        hdr.srv6_sid1.segment_id = s1;
        hdr.srv6_sid2.setValid();
        hdr.srv6_sid2.segment_id = s2;
        hdr.ipv6_outer.dst_addr = s2;
        build_srv6(2);
    }
```
The TEID field has 32 bits lenght

happyPisces, 2020-11-14:
Hi, I have been trapped with this problem too, have you solved this problem now?

guimvmatos, 2020-11-14:
> 
> 
> Hi, I have been trapped with this problem too, have you solved this problem now?

HI @happyPisces, 

I haven't solve it yet, but I discover that If I comment on one of the lines below, it works. I don't know why this is happening.
```
srv6_t_insert_2
srv6_t_insert_3 
```
Tell me if it works for you. And if you find out the cause of this behavior, please, tell me too.

happyPisces, 2020-11-16:
HI @guimvmatos ,
Thanks. If I find out something, I will tell you.

happyPisces, 2020-11-26:
We solved the problem by fixing our command.txt,
The wrong command.txt is as follows:

The correct command.txt is as follows:
table_cam_add_entry forward set_output_port 0b0000 => 0b00000001
table_cam_add_entry forward set_to_cpu 0b0001 => 0b00000010

I think the number of ancitons in command.txt should match the number of acrtions in table,
So you should add one more tcam entry whose action is srv6_t_insert_3, or you can circumvent this problem by commenting srv6_t_insert_3 in your table just as you did before.

发送自 Windows 10 版邮件应用

发件人: Guilherme Matos
发送时间: 2020年11月15日 5:11
收件人: NetFPGA/P4-NetFPGA-public
抄送: happyPisces; Mention
主题: Re: [NetFPGA/P4-NetFPGA-public] ERROR: not enough fields provided tocomplete _hexify() (#56)

Hi, I have been trapped with this problem too, have you solved this problem now?
HI @happyPisces,
I haven't solve it yet, but I discover that If I comment on one of the lines below, it works. I don't know why this is happening.
srv6_t_insert_2
srv6_t_insert_3 
Tell me if it works for you. And if you find out the cause of this behavior, please, tell me too.

XiaoM1573, 2020-11-27:
This problem i have solved. You should add some zero in your commands.txt.
For table teid_exact, actually it has five parameters and their order is fixed.
i.e.
table_cam_add_entry teid_exact insert_srv6_list_2 0x20010001000000000000000000000002 => 0x00A10000000000000000000000000002 0x20010001000000000000000000000002 0 0 0

i.e.
table_cam_add_entry teid_exact insert_srv6_list_3 0x20010001000000000000000000000002 => 0 0 0x00A10000000000000000000000000002 0x00A10000000000000000000000000006 0x20010001000000000000000000000002