NetFPGA/P4-NetFPGA-public #60: [HELP] Switch Calculator Tutorial (Step 9 & 10)

mharris-t, 2021-03-26:
Hi,

**My system info:**

LSB Version: core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID: Ubuntu
Description: Ubuntu 20.04 LTS
Release: 20.04
Codename: focal

Python env ver: 2.7

Board: NetFPGA Sume Board
Project: P4 NetFPGA Project

Xilinx Version: 2020.2

SDNet Version: 2018.2

I am using the NetFPGA Sume development board to understand its technicalities and work flow with P4. I am using this tutorial to program the switch calculator in the board. I am stuck in step 9, where I need to run the simulation with the GUI. I get the following error when I run the command:
`
 ./tools/scripts/nf_test.py sim --major switch --minor default --gui`

[image]

Before this error appeared, I copied the input arbiter folder from the P4 repo to /tools/Xilinx/Vivado/2020.2/data/ip/xilinx/ and made the changes to vv_index.xml file as follows:

```
	<IP>
		<VLNV value="NetFPGA:NetFPGA:input_arbiter:*">
		</VLNV>
		<DisplayName value="NetFPGA Input Arbiter">
		</DisplayName>
		<HideInGui value="false">
		</HideInGui>
		<!-- <CoreRevision value="4">
		</CoreRevision> -->
		<Description value="NetFPGA Input Arbiter">
		</Description>
		<ComponentPath value="xilinx/input_arbiter_drr_v1_0_0/component.xml">
		</ComponentPath>
		<Taxonomies>
			<Taxonomy value="na">
			</Taxonomy>
		</Taxonomies>
		<!-- <AutoFamilySupport value="Level_2">
		</AutoFamilySupport> -->
		<SubcoreVLNVs>
			<SubcoreVLNV value="xilinx.com:ip:axi_lite_ipif:3.0">
			</SubcoreVLNV>
		</SubcoreVLNVs>
		<VendorURL value="http://www.xilinx.com">
		</VendorURL>
	</IP>
```

Any help in resolving this matter will be beneficial for me.

salvatorg, 2021-03-26:
Important: This project uses Vivado 2018.2. You will likely run into issues if you do not use this version of Vivado. Also, make sure to check the box indicating that you would like to install the SDK when installing Vivado.

[Source](https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Getting-Started#steps-for-initial-setup)

mharris-t, 2021-03-26:
Thank you very much for the prompt reply! I see, if that's the case, what's the appropriate Ubuntu OS for Vivado 2018.2?

salvatorg, 2021-03-29:
For compatibility refer to Xilinx's website. I assume that you can install Vivado 2018.02 in Ubuntu 20.04.

mharris-t, 2021-04-07:
I was able to resolve the issues. Basically, I had to format my workstation and install Ubuntu 16.04.3 LTS. since Vivado 2018.2 supports only Ubuntu 16.04.3. I had to circumvent with a lot issues surrounding some outdated packages. However, I was able to make everything work! 

P.S - I am not sure if Ubuntu 2020.04 would support Vivado 2018.2. If anyone is able to make it work, please do comment here and let us know!

closed 2021-04-07 by mharris-t