NetFPGA/P4-NetFPGA-public #61: Error while running P4-SDNet compiler for tcp monitor assignment

MarioPatetta, 2021-04-07:
Hello,

I'm trying to run the tcp tutorial assignment and, when I run 

> $ cd $P4_PROJECT_DIR && make

I get this error:

> make[1]: Entering directory '/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/tcp_monitor/testdata'
> ./gen_testdata.py
> Traceback (most recent call last):
>   File "./gen_testdata.py", line 150, in <module>
>     trace = mix_flows([flow1, flow2, flow3])
>   File "./gen_testdata.py", line 143, in mix_flows
>     trace = map(next, random.sample([iter(trace)]*len(trace) + [iter(flows[fid])]*len(flows[fid]), len(trace)+len(flows[fid])))
> TypeError: object of type 'itertools.imap' has no len()
> Makefile:34: recipe for target 'all' failed
> make[1]: *** [all] Error 1
> make[1]: Leaving directory '/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/tcp_monitor/testdata'
> Makefile:60: recipe for target 'frontend' failed
> make: *** [frontend] Error 2

I've already tried starting again from scratch multiple times, making sure to follow each step but the result is still the same.

Let me know if I can give you more information in order to make my situation clearer. Thank you so much for the support.

salvatorg, 2021-04-07:
Maybe you want to quick check if u are using the correct python(2,3) version?

MarioPatetta, 2021-04-07:
> Maybe you want to quick check if u are using the correct python(2,3) version?

Thanks for the answer!

I am using python 2.7.17. I think it should be fine, right?

salvatorg, 2021-04-08:
I believe its the correct one..
Is there any previous error maybe?

MarioPatetta, 2021-04-08:
There are no previous error, also because this is the first step of the assignment to run through the command line.

I had already went through this assignment before and didn't get this error. I am now going back to it because I needed to start back from scratch but I find myself stuck...

MarioPatetta, 2021-04-21:
I managed to solve the issue. 
It was due to the fact that the map operator apparently was changed in python 2 in order to return an iterator instead of a list.

You can find the fix [here](https://devtut.github.io/python/map-function.html#basic-use-of-map-itertools-imap-and-future-builtins-map).

It consists in simply apply list() to the result of map()

closed 2021-04-21 by MarioPatetta