NetFPGA/P4-NetFPGA-public #62: Cannot Push Bitfiles into NetFPgA SUME - No such device!!!

mharris-t, 2021-04-08:
Hi,

**System specs:**

Motherboard: Fujitsu D3227-A1
CPU: Intel(R) Xeon(R) CPU E3-1230 v3 @ 3.30GHz

**Operating System:**

Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

**Vivado Version:** 2018.2
**SDNet Version:** 2018.2

I am having issues with pushing bitfiles into the netfpga. Moreover, I am having trouble in detecting the interfaces nf0, nf1, nf2 and nf3. However, my motherboard has been able to detect the netfpga and I have been able to connect to the board with USB over minicom. 

[image]

[image]

I have rebooted my system several times. Restarted my networking services.  Reinstalled **sume_riffa** drivers. But these don't seem to work. The _**lspci**_ command does not display the netfpga in the results. But **_djtgcfg enum_** command shows the detected NIC in the system. Also, Vivado's Hardware Manager detects the Xilinx chipset. 

[image]

[image]

[image]

aomtoku, 2021-04-08:
13 of https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Tutorial-Assignments#what-to-do would be helpful for you.

mharris-t, 2021-04-08:
@aomtoku, I am stuck at exactly that point at step 13. When I try running running the bash script **program_switch.sh,** this is what happens:

[image]

sume_riffa driver is loaded using modprobe, but the netfpga interfaces do not appear even after restarting the interfaces configuration in /etc/network/interfaces.

aomtoku, 2021-04-08:
did you do this  of 13? 
==================================================================================
Note: Make sure that all the configuration writes succeed. If this is the first time you are programming the FPGA since the machine was last powered off then you might get ERROR messages stating No such device.

To resolve this, follow the below-mentioned steps:

a. Reboot the Machine

b. Execute the following commands (taken from Step 9 on Getting Started Page in this repo):

[root@nf-test109 ~]# cd $DRIVER_FOLDER

[root@nf-test109 sume_riffa_v1_0_0]# make all

[root@nf-test109 sume_riffa_v1_0_0]# make install

[root@nf-test109 sume_riffa_v1_0_0]# modprobe sume_riffa

[root@nf-test109 sume_riffa_v1_0_0]# lsmod | grep sume_riffa

c. Now, you will be able to see nf0, nf1, nf2 and nf3 ports in your ifconfig output. Re-execute the aforementioned commands in Step 13. This time the configuration writes & the loading bitstream process should succeed.

mharris-t, 2021-04-08:
Yes, I have rebooted my machine several times and loaded the sume_riffa drivers. Even then, the interfaces do not appear unfortunately.

aomtoku, 2021-04-08:
What is the result of lspci for sume?

mharris-t, 2021-04-08:
Neither Xilinx nor sume gets detected in lspci. But I have been able to test the sume nic using minicom and it is detected in Vivado's hardware manager.

aomtoku, 2021-04-08:
I would suggest 
1) check PCIe connectivity 
2) check writing bitifile

mharris-t, 2021-04-08:
Alright, it seems like my NIC isn't being detected in the PCIe slot. I will see what I can do. Could be a loose connection or the NIC is incompatible with my motherboard. Thank you very much!

closed 2021-04-08 by mharris-t

MyLifeForViolet, 2023-11-16:
# @mharris-t I meet the same problem with you.Have you ever solved this? I have tried all solutions mentioned above,unfortunately failed.

MyLifeForViolet, 2023-11-16:
@mharris-t  Hi.I meet the same problem with you.Have you ever solved this? I have tried all solutions mentioned above,unfortunately failed.