NetFPGA/P4-NetFPGA-public #64: Xilinx SDNet not found

Carmine50, 2021-04-15:
Hi,

I am having trouble to download Xilinx SDNet. Do you know if there is any other way apart from asking to a local sales managet of Xilinx to get the this tool?

Thanks

salvatorg, 2021-04-16:
As far I am aware Xilinx is about to release a new SDNet at some point in the future and they have removed the download link for the old version from their website.
You might want to ask around the community for the old version of the SDNet.

Carmine50, 2021-04-16:
@salvatorg do you know by any chance when it will be available?

mharris-t, 2021-04-19:
I had to contact Xilinx representative to obtain the download link.

salvatorg, 2021-04-19:
> @salvatorg do you know by any chance when it will be available?

@Carmine50 sorry, I have no idea.

cathalmccabe, 2021-12-21:
Hi all, if you need access to SDNet please contact xup@xilinx.com or make a donation request on www.xilinx.com/university