NetFPGA/P4-NetFPGA-public #65: Add Wave error during Sume Simulation

MarioPatetta, 2021-04-24:
Hello!

When I run the command `$ cd $SUME_FOLDER && ./tools/scripts/nf_test.py sim --major switch --minor default` to run the Sume simulation I get this error:

> ERROR: [Wavedata 42-471] Note: Nothing was found for the following items: top_tb/top_sim/nf_datapath_0/nf_sume_sdnet_wrapper_1/inst/SimpleSumeSwitch_inst/const_reg_rw_0
> ERROR: [Common 17-39] ‘add_wave’ failed due to earlier errors.

I have tried manipulating both my p4 source code and my gen_testdata.py script but the result is always the same.

I have also found [here](https://pub.tik.ee.ethz.ch/students/2019-FS/MA-2019-19.pdf) and [here](https://gitlab.ethz.ch/nicosc/master-thesis/-/blob/master/doc/plan.org) that this is a known error, but I couldn't find any tips on how to solve it.

Does somebody have any advice?

rst0git, 2021-04-24:
Hi @MarioPatetta, Could you please try with the following script?

```sh
source tools/settings.sh

# Build Vivado core IP modules
cd $SUME_FOLDER/lib/hw/xilinx/cores/tcam_v1_1_0/ && make update && make
cd $SUME_FOLDER/lib/hw/xilinx/cores/cam_v1_1_0/ && make update && make
cd $SUME_SDNET/sw/sume && make
cd $SUME_FOLDER && make

# Build sume_riffa driver
make -C $DRIVER_FOLDER all
make -C $DRIVER_FOLDER install
modprobe sume_riffa

# Generate verilog code and API/CLI tools
make -C $P4_PROJECT_DIR

# Run SDNet simulation
cd $P4_PROJECT_DIR/nf_sume_sdnet_ip/SimpleSumeSwitch
./vivado_sim.bash

# Generate scripts for NetFPGA SUME simulation
make -C $P4_PROJECT_DIR config_writes
make -C $P4_PROJECT_DIR install_sdnet
make -C $NF_DESIGN_DIR/test/sim_switch_default

# Run NetFPGA simulation
cd $SUME_FOLDER
./tools/scripts/nf_test.py sim --major switch --minor default
```

MarioPatetta, 2021-04-26:
Hello @rst0git, and thanks for answering. 

I have tried your script and it sadly resulted in the same exact error.

MarioPatetta, 2021-04-27:
I managed to solve the problem. It was due to my mistake. 
Tank you for your time!

closed 2021-04-27 by MarioPatetta