NetFPGA/P4-NetFPGA-public #66: Issue trying to use a lpm LUT

MarioPatetta, 2021-05-18:
Hello!

I am trying to use a lpm lut to count the number of leading zeros of a binary number.

The lines of  the commands.txt file that I use to fill the entries of the table are as follows:

```
table_lpm_add_entry lookup_rarity set rarity 0.0.0.0.0.0.0.1/1 => 0b000001
table_lpm_add_entry lookup_rarity set rarity 0.0.0.0.0.0.0.1/2 => 0b000010
table_lpm_add_entry lookup_rarity set rarity 0.0.0.0.0.0.0.1/3 => 0b000011
```

And so forth.

My idea is to compare the 64 bit number with the number made of all 0s and a 1 as LSB and output the length of the longest prefix matching. I've used the dot separated format for the prefix as suggested in line 279 of p4_px_tables-py in P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/bin.

When I run the P4-SDNet compiler I get this error:

>   File "/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/bin/p4_px_tables.py", line 438, in <module>
>     main()
>   File "/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/bin/p4_px_tables.py", line 435, in main
>     write_px_tables()
>   File "/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/bin/p4_px_tables.py", line 424, in write_px_tables
>     table.write_px_file()
>   File "/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/bin/p4_px_tables.py", line 222, in write_px_file
>     fout.write("{} {**:d**} {:X}\n".format(prefix, length, value))
> ValueError: Unknown format code 'd' for object of type 'str'

So I went at line 222 of p4_px_tables.py and removed the :d from inside the curly brackets (bolded in the error report). That seemed to solve the issue, in fact the P4-SDNet compiler runs smoothly and I get the lookup_rarity.tbl in the src directory.

I've read in the [Workflow Overview](https://github.com/NetFPGA/P4-NetFPGA-public/wiki/Workflow-Overview#workflow-steps) that, when working with lpm tables, the compiler should generate a bash script that must be run before the SDNet simulation and I actually found one named 

> lookup_rarity_create_dat_file_for_sim.bash

located in projects/<project_name>/nf_sume_sdnet_ip/SimpleSumeSwitch

When I run it I get:

> +java -jar Testbench/lpm_cfg.jar lookup_rarity.tbl -o lookup_rarity.dat -k 64 -v 8 -l 6
> PROGRAM BEGIN : 1 ms, +1621260240966
> PREFIX LOADED : 17 ms, +16
> Warning: no. prefixes == 64 > (no. key slot)/2 == 2^6-1
> BOUNDS CREATED: 17 ms, +0
> BOUNDS SORTED : 18 ms, +1
> RANGES CREATED: 18 ms, +0
> LEVELS CREATED: 19 ms, +1
> 
> cfg_word size:74
> log2MaxStages:3
> log2MaxStageNodes:7
> log2MaxWords:2

After that, I run the SDNet simulation and I get these errors:

> Running: /opt/Xilinx/Vivado/Vivado/2018.2/bin/unwrapped/lnx64.o/xelab -L work --debug all -sv_lib dpi.so SimpleSumeSwitch_tb glbl 
> Multi-threading is on. Using 30 slave threads.
> Starting static elaboration
> ERROR: [VRFC 10-426] cannot find port tuple_out_rarity_cnt1_reg_rw_output_1_DATA on this module [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1420]
> ERROR: [VRFC 10-426] cannot find port tuple_out_rarity_cnt1_reg_rw_output_1_VALID on this module [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1419]
> ERROR: [VRFC 10-426] cannot find port tuple_in_rarity_cnt1_reg_rw_input_1_DATA on this module [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1418]
> ERROR: [VRFC 10-426] cannot find port tuple_in_rarity_cnt1_reg_rw_input_1_VALID on this module [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1417]
> ERROR: [VRFC 10-2063] Module <rarity_cnt1_cpu_regs> not found while processing module instance <rarity_cnt1_cpu_regs_inst> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/rarity_cnt1_reg_rw_2_t.HDL/rarity_cnt1_reg_rw_2_t.v:181]
> ERROR: [VRFC 10-2063] Module <TopPipe_lvl_4_t> not found while processing module instance <TopPipe_lvl_4> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1445]
> ERROR: [VRFC 10-2063] Module <rarity_cnt0_reg_rw_2_t> not found while processing module instance <rarity_cnt0_reg_rw_2> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1489]
> ERROR: [VRFC 10-2063] Module <TopPipe_lvl_5_t> not found while processing module instance <TopPipe_lvl_5> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1520]
> ERROR: [VRFC 10-2063] Module <rarity_cnt0_reg_rw_1_t> not found while processing module instance <rarity_cnt0_reg_rw_1> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1564]
> ERROR: [VRFC 10-2063] Module <TopPipe_lvl_6_t> not found while processing module instance <TopPipe_lvl_6> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1595]
> ERROR: [VRFC 10-2063] Module <rarity_cnt0_reg_rw_0_t> not found while processing module instance <rarity_cnt0_reg_rw_0> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1639]
> ERROR: [VRFC 10-2063] Module <TopPipe_lvl_7_t> not found while processing module instance <TopPipe_lvl_7> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1670]
> ERROR: [VRFC 10-2063] Module <rarity_cnt1_reg_rw_0_t> not found while processing module instance <rarity_cnt1_reg_rw_0> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1714]
> ERROR: [VRFC 10-2063] Module <TopPipe_lvl_8_t> not found while processing module instance <TopPipe_lvl_8> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1745]
> ERROR: [VRFC 10-2063] Module <ipv4_forward_t> not found while processing module instance <ipv4_forward> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1789]
> ERROR: [VRFC 10-2063] Module <rarity_cnt1_reg_rw_1_t> not found while processing module instance <rarity_cnt1_reg_rw_1> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1820]
> ERROR: [VRFC 10-2063] Module <TopPipe_lvl_9_t> not found while processing module instance <TopPipe_lvl_9> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1851]
> ERROR: [VRFC 10-2063] Module <TopDeparser_t> not found while processing module instance <TopDeparser> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1885]
> ERROR: [VRFC 10-2063] Module <S_BRIDGER_for_hash_lrc_0_tuple_in_hash_lrc_input> not found while processing module instance <S_BRIDGER_for_hash_lrc_0_tuple_in_hash_lrc_input> [/home/mario/mario_test/P4-NetFPGA-live/contrib-projects/sume-sdnet-switch/projects/mario/nf_sume_sdnet_ip/SimpleSumeSwitch/SimpleSumeSwitch.v:1927]
> INFO: [#UNDEF] Sorry, too many errors..
> ERROR: [XSIM 43-3322] Static elaboration of top level Verilog design unit(s) in library work failed.

At this point I am stuck because I don't really get what I did wrong. Could somebody help me?

rst0git, 2021-05-20:
>ValueError: Unknown format code 'd' for object of type 'str'

@MarioPatetta  Could you try to replace `p4_px_tables.py` with the following version?
https://github.com/mtpsa/P4-NetFPGA-MTPSA/blob/mtpsa/contrib-projects/sume-sdnet-switch/bin/p4_px_tables.py

MarioPatetta, 2021-05-22:
Hello @rst0git and thank you for the answer. 
I have tried what you suggest but I get the exact same errors.

Also, I have opted for a different solution to my need to compute the number of leading zeros. In fact, I now do as they do in this paper (they determine the position of the first 1, that's basically the same thing): 

https://boa.unimib.it/retrieve/handle/10281/273149/399451/2020_NOMS_Log_Exp_Entropy_P4.pdf

This way I don't have the need to use a lookup table. The problem is that I still get the same errors. This probably means that the errors i am getting are uncorrelated to the use of a lpm LUT.

Do you have any suggestion on how to find out what is creating the problem?

rst0git, 2021-05-22:
@MarioPatetta Could you try the following?
```
cd ~/
git clone https://github.com/rst0git/P4-NetFPGA-IPv4-LPM
cd P4-NetFPGA-IPv4-LPM
source tools/settings.sh
./run.sh
```
If this works, you can try to uncomment the build steps in the `run.sh` file.

MarioPatetta, 2021-05-24:
@rst0git I've run the code you sent me and it went through without errors. I then uncommented the last steps of the `run.sh` and got this error:

```
make: Leaving directory '/home/mario/P4-NetFPGA-IPv4-LPM/contrib-projects/sume-sdnet-switch/projects/learning_switch'
make: *** /home/mario/P4-NetFPGA-IPv4-LPM/contrib-projects/sume-sdnet-switch/projects/learning_switch/simple_sume_switch/test/sim_switch_default: No such file or directory.  Stop.
=== Error: No tests match sim_switch_default - exiting
```

I still think that at this point the problem is not linked to the use of a lpm LUT since I'm having the same errors even in my other version of the project that doesn't use it.

MarioPatetta, 2021-05-27:
I have figured out that the problem is indeed linked to part of my p4 code, not to the use of the LUT.
In fact, the error disappears if I comment out this section:

```
// alternate between the two registers
reg_track_reg_raw(index,0,1,REG_ADD,reg_track);					        // since the register entry is just 1 bit long, reg_track will only be either 0 or 1

if (reg_track == 1) {
	RarityCnt0_reg_rw(index,0,REG_READ,prev_rarity);			        // read the rarity stored in the previous register
	if (pkt_rarity > prev_rarity) {						        // compare it to the current rarity
		RarityCnt1_reg_rw(index,pkt_rarity,REG_WRITE,stored_rarity);
	}									        // write the highest in the actual register
	else if (pkt_rarity <= prev_rarity){
		RarityCnt1_reg_rw(index,prev_rarity,REG_WRITE,stored_rarity);
	}
}
else if (reg_track == 0) {
	RarityCnt1_reg_rw(index,0,REG_READ,prev_rarity);
	if (pkt_rarity > prev_rarity) {
		RarityCnt0_reg_rw(index,pkt_rarity,REG_WRITE,stored_rarity);
	}
	else if (pkt_rarity <= prev_rarity) {
		RarityCnt0_reg_rw(index,prev_rarity,REG_WRITE,stored_rarity);
	}
}
```

But I don't really see why this code shouldn't work. Do you have any idea?

MarioPatetta, 2021-05-27:
Actually, commenting out some lines at a time I figured out that the error occurs when I have a situation like this:

```
if (something) {
    access to a register
}
else if (someting else) {
    access to the same register
}
```

I know that in p4 you can only access registers once per processed packet, but in this case the use of if-else should ensure that I don't try to access it more than once, right?

closed 2021-07-06 by MarioPatetta