NetFPGA/P4-NetFPGA-public #67: CLI tool not working as expected

MarioPatetta, 2021-10-05:
Hello!

I have been experiencing some issues with the CLI, both with the project I am working on and the switch_calc tutorial. 

After programming the FPGA (thus passing both SDNet and SUME simulations), some commands of the tool work and some others have unexpected results:

- list_(t)cam_tables and list_regs work just fine
- table_(t)cam_add_entry results in an error that shuts the CLI. Example:

```
>> table_cam_add_entry ipv4_forward set_dst_port 08:11:11:11:11:08 => 0b00000001
CAM_Init_ValidateContext() - done
WROTE 0x44020050 = 0x11111108
WROTE 0x44020054 = 0x0811
WROTE 0x44020080 = 0x0101
python: ioctl: Unknown error 512
```

- The exact same error occurs with table_(t)cam_read_entry and table_(t)cam_get_size and basically any other command involving cam or tcam tables. I didn't try with lpm tables since they are not present in neither my project or switch_calc.
- reg_write does not produce any errors but reg_read always returns -559038737 for every index of every register, even if I have previously wrote something else in that spot.

Does anyone have any idea on how to address this problem?

salvatorg, 2021-11-04:
-559038737 = 0xdeadbeaf
this is an indication of something bad (u already know this) happens with the registers.

MarioPatetta, 2021-11-26:
I am looking for a different solution to read from the registers. I have tried building a custom serial interface modifying the helloworld.c file in `simple_sume_switch/sw/embedded/src`. 

Do you have any suggestions on how to read from the registers that I instantiate in the p4 code? I have tried to use `Xil_In32()`, passing as input `base_addr + addr_offset` that I found in `sw/CLI/SimpleSumeSwitch_extern_defines` but the output that I get is DEC0DE1C, which should mean that the address is not recognised, right?

Every suggestion is much appreciated.