NetFPGA/P4-NetFPGA-public #68: Is code request form still available?

kyeot95, 2022-01-25:
I''m sorry for uploading this question in [issues].
I send code access request form about two weeks ago
and I still got no reply.
I also send emails to contact netfpga and got no reply either.
Is code request form still available?
or am I just been rejected to code access?

salvatorg, 2022-02-09:
Apologies for the delay, your request will be processed asap.

closed 2022-02-09 by salvatorg

kyeot95, 2022-03-24:
Did I took it wrong..?
I understand the access permission is about  [https://github.com/NetFPGA/NetFPGA-SUME-live].

and No. I still don't have permission to above site.
what is happening?

salvatorg, 2022-03-31:
There are two separate repos for NetFPGA-SUME and P4-NetFPGA which have 2 separate registration forms.
In any case the invitation to access NetFPGA-SUME-live has been sent to you.